UniversalRobots/Universal_Robots_Client_Library #28: I can't cmake ,please help me.

aoligeihahah, 2020-11-22:
/home/he/catkin_ws/src/Universal_Robots_Client_Library/include/ur_client_library/rtde/data_package.h:32:19: fatal error: variant:There is no such file or directory

aoligeihahah, 2020-11-22:
[image]

fmauch, 2020-11-22:
Please give us a little bit more information:
 - On what kind of system are you on?
 - Which compiler are your using?
 - Which branch of this library are you trying to build?

Also please give the full build output using a [code block](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code), not a screenshot.

aoligeihahah, 2020-11-23:
sorry.
My system is:
ubantu16.04+ros-kinetic
My code is:
mkdir -p catkin_ws/src && cd catkin_ws
git clone https://github.com/UniversalRobots/Universal_Robots_Client_Library.git src/Universal_Robots_Client_Library
sudo apt update -qq
rosdep update
rosdep install --from-paths src --ignore-src -y
catkin_make_isolated
source devel_isolated/setup.bash

gavanderhoorn, 2020-11-23:
> My system is:
ubantu16.04+ros-kinetic

Does that support C++17?

aoligeihahah, 2020-11-23:
How to check whether the  my system supports

aoligeihahah, 2020-11-23:
> How to check whether the my system supports

ok，It should be the problem.I tried get the system to support re installation.

fmauch, 2020-11-23:
You can use the boost branch of the library that will not require c++17.

fmauch, 2020-11-24:
It would still be nice if you could post your full cmake output as in theory we have a check for c++17 in place that should result in a cmake error already.

aoligeihahah, 2020-11-24:
Thank you. I installed gcc-9.1.0, which solved this problem. But I had a problem. I succeeded in roslaunch example_ organization_ Ur_ Launch ex-ur10-1.launch, but catkin_ make_ Isolated has the error shown in the figure below. I want to get help.

my code：
  cd catkin_ ws && source devel/ setup.bash
  catkin_ make_ isolated

Create an example_organization_ur_launch : 
catkin_ create_ pkg example_ organization_ Ur_ launch ur_ client_ library

[image]
[image]

fmauch, 2020-11-24:
What do you want to achieve? If you want to build the `ur_robot_driver` then I suggest you follow the [`Instructions`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#alternative-all-source-build)

And please do not post screenshots of shell output, but use code blocks as mentioned above.

fmauch, 2020-11-24:
> It would still be nice if you could post your full cmake output as in theory we have a check for c++17 in place that should result in a cmake error already.

Looks like gcc5 supports some C++17 features already and therefor doesn't trigger the check. However, `std::variant` is only supported from gcc7 up. I updated the documentation in #29 accordingly.

gavanderhoorn, 2020-11-25:
Which OS do you want to support here @fmauch?

I've found `target_compile_features(.. PUBLIC cxx_std_17)` to work quite reliably, but it does need a relatively up-to-date CMake.

An additional advantage of using `target_compile_features(..)` is that it is cross-platform. I believe the currently used macro is not.

fmauch, 2020-11-25:
> Which OS do you want to support here @fmauch?
>
> I've found `target_compile_features(.. PUBLIC cxx_std_17)` to work quite reliably, but it does need a relatively up-to-date CMake.
>
> An additional [...]

Quickly skimming over the docs this seems indeed to be the better approach. Especially since the minimum required cmake version would be 3.1 which should be available on all target platforms (with ubuntu 16.04 being the minimal one atm.) However, I assume that this still covers the same issues with `std::variant` and gcc5.

gavanderhoorn, 2020-11-25:
> However, I assume that this still covers the same issues with `std::variant` and gcc5.

I don't understand what you mean by this.

`target_compile_features(..)` checks for features, not versions. So if you specify support for `std::variant` must be present, it will check for that, instead of just assuming it will be there if a compiler accepts `--std==c++17`.

Whether there is a check for `std::variant` I don't know, you'd have to check.

fmauch, 2020-11-25:
It doesn't look like there is a specific check for `std::variant` (https://cmake.org/cmake/help/latest/prop_gbl/CMAKE_CXX_KNOWN_FEATURES.html#prop_gbl:CMAKE_CXX_KNOWN_FEATURES)

So, everything we could do is still to check for c++17 support (in general). In fact, `cxx_std_17` is only available from cmake 3.8 upwards.

gavanderhoorn, 2020-11-25:
Hm.

Well, three options:

 1. use `boost::variant` on systems without C++17 support (if `std::variant` is the reason you're targetting C++17)
 1. use an alternative `variant` implementation (such as [mpark/variant](https://github.com/mpark/variant))
 1. upgrade CMake as part of the build setup

It's actually not that difficult to upgrade CMake on any Linux OS, as it's distributed via `pip` these days. A `pip install --user cmake` will install CMake `3.18.4` for the current user, without touching the system installed one.

All invocations of `cmake` for *that* user will see the `pip` installed one.

It's not very nice, but I've used it in the past to support old systems with really out-of-date CMake versions (and CMake `3.5.1` is getting really old, so is GCC 5).

fmauch, 2020-11-25:
Options 1 and 2 would hurt our requirement to be independent of external libraries (which is basically why we have the boost branch). Option 3 also doesn't look too fancy to me, at least for automated installation... We could require higher versions for cmake and gcc though and hint users how to upgrade on their older systems. For those who don't want to do that, there's still the boost branch that uses option 1.

> With Xenial end-of-standard-support being only 5 months away (Apr 2021), you could also decide to stick with what you have now until that time.

I'm thinking the same about this issue... Especially as with #29 this will hopefully be documented well enough.

gavanderhoorn, 2020-11-25:
> Options 1 and 2 would hurt our requirement to be independent of external libraries

To be fair though: there are already dependencies on "external" dependencies. You've just decided to vendor them in.

fmauch, 2020-11-25:
> To be fair though: there are already dependencies on "external" dependencies. You've just decided to vendor them in.

Fair enough :-)

closed 2020-11-30 by aoligeihahah