UniversalRobots/Universal_Robots_Client_Library #30: RTDE protocol not used?

yottasyno, 2021-02-01:
The [scriptfile.urscript](Universal_Robots_Client_Library/examples/resources/scriptfile.urscript) seems to only use the function socket_read_binary_integer(nbytes, socket) to get the desired joint angles which will be set with the function servoj(q, v, a, dt, look_ahead_time, gain). I cannot see at which point this driver is using the [RTDE protocol](https://www.universal-robots.com/articles/ur/real-time-data-exchange-rtde-guide/).

1. **Is the protocol not used in this driver?** In [data_package.cpp](Universal_Robots_Client_Library/src/rtde/data_package.cpp) on the other hand the registers are defined and there are input and output recipes in Universal_Robots_Client_Library/examples/resources/ .

2. **Does anyone have an example of how to use the servoj(...) command by using the RTDE protocol?**

Thanks a lot for your help!

t-schnell, 2021-02-01:
The RTDE protocol is only used for reading data and setting inputs in this driver. Specific data for motion control, such as the desired joint angles for servoj are, as you noticed, sent via a separate socket connections.

yottasyno, 2021-02-01:
Thanks a lot!

closed 2021-02-01 by yottasyno