UniversalRobots/Universal_Robots_Client_Library #31: Placement of control urscript and RTDE recipe

t-schnell, 2021-02-04:
Currently, the urscript file to be run on the robot is intended to be placed with the external driver, such as the UR ROS Driver or the minimal example. While this script offers the robot side counterpart of the driver-level control loop, it directly interfaces with the functions and values of the client library. This is noticeable in the reverse interface functions, but maybe even more critically in the string replacements and the ControlMode enum, as these have to be manually be kept in synch across repositories.

In a similar manner, the RTDE based communication requires, at least in the writing component, to have certain values set in the recipe file.

Possible ideas to move around this would be:
- move these files (control.urscript and rtde_input_recipe.txt) to the client library
- split files (easy for recipe, but could also work for control script)

fmauch, 2021-02-05:
In principle, the files in the two repositories don't necessarily have to be kept in sync. While the one here merely serves as an example, the one in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver would be the specific file used for ROS control.

Things are a bit wildly connected with each other, as there are certain library components that require certain script code to be running (e.g. the [ReverseInterface](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/include/ur_client_library/comm/reverse_interface.h)) and also the mentioned [script replacement](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/ur/ur_driver.cpp#L90)

I think, it might be beneficial to strip things out altogether.

E.g. script replacement could be handled in a separate class so that the `UrDriver` class gets a functioning script string instead of loading it from a file.

For this, it might be beneficial to work out which core components we want to have in a "script" in either case and then document those requirements. Anything going beyond this should be optional and documented, as well.

Is there any specific use case in mind leading up to this request?

t-schnell, 2021-02-17:
Current plan:

- rtde_output_recipe 
    - stays in client, not required by library

- rtde_input_recipe 
    - minimal recipe in library, but allow overwrite to avoid conflicts
    - add output for "in use" case so user is informed of conflict

- control script
    - move to library
    - optional overwrite by client

This way ease of use for clients is increased and required duplication between clients minimized.

fmauch, 2021-02-17:
@t-schnell I would rephrase "client" with "application", as we have the "client library" with the "ROS driver" and the "IISAC driver" as applications.

Also, I think for the input recipe we planned to not necessarily have the minimal configuration inside a recipe file, but have it inside the source code and add a flag whether IO functionality should be enabled or not.

fmauch, 2024-09-11:
I think, this can be closed by now.

closed 2024-09-11 by fmauch