UniversalRobots/Universal_Robots_Client_Library #45: Terminate called without an active exception

urmahp, 2021-03-12:
# Summary

When deleting the UrDriver object the following error is printed `terminate called without an active exception`. 

This happens when stopping the ROS driver, where it happens at following [line](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface_node.cpp#L39), which deletes the hardware interface and therefore also the UrDriver object.

The same error occurs in the Isaac driver when deleting the UrDriver object or when creating my own instance and then deleting it again. 

# Impact
The program is terminated when deleting the UrDriver, meaning that code after deleting the driver will not get executed. 

# Issue details
Running following code piece will produce the error. 
```C++
// We need a callback function to register. See UrDriver's parameters for details.
void handleRobotProgramState(bool program_running)
{
  std::cout << "\033[1;32mProgram running: " << std::boolalpha << program_running << "\033[0m\n" << std::endl;
}

std::unique_ptr<UrDriver> g_my_driver;
int main(int argc, char* argv[])
{
  std::unique_ptr<ToolCommSetup> tool_comm_setup;
  g_my_driver.reset(new UrDriver(ROBOT_IP, SCRIPT_FILE, OUTPUT_RECIPE, INPUT_RECIPE, &handleRobotProgramState, false,
                                 std::move(tool_comm_setup), CALIBRATION_CHECKSUM));
  g_my_driver.reset(); // without a call to reset, will result in the same error
  return 0;
}
```
Running the code resolves in the following error:
```
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /home/mahp/ur_driver_ws/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h
Warning: No realtime capabilities found. Consider using a realtime system for better performance
         at line 396 in /home/mahp/ur_driver_ws/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h
terminate called without an active exception
Aborted (core dumped)

```
I have tracked the issue to be the [thread](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/ur/ur_driver.cpp#L204-L236) running in UrDriver and also the [thread](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/include/ur_client_library/comm/script_sender.h#L78-L90) running in the scriptSender. These two threads are not joined or detached when deleting the object, which is leading to the error `terminate called without an active exception`.

# Possible solution
It is not possible to stop the running threads and join them, because of the following [line](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/comm/server.cpp#L91) in server.cpp. Accept is a blocking call meaning that the threads are stuck here until a robot connects to the socket. Meaning that a join call could hang for a long time before the thread is returned depending on when the robot is connecting to the socket.  

A possible solution is to detach the threads in the destructor, when destroying the script_sender and the UrDriver object. The problem with detaching the threads is that the threads will still live after the object gets destroyed causing some issues if you create a new object right after. 

A solution with detaching the thread in the scriptSender could look like this, some similar changes could be applied in the UrDriver class. 
```c++
  ~ScriptSender()
  {
    stop_thread_ = true;
    server_.disconnectClient();
    script_thread_.detach();
  }
  void runScriptSender()
  {
    while (!stop_thread_)
    {
      if (!server_.accept())
      {
        throw std::runtime_error("Failed to accept robot connection");
      }
      if (requestRead())
      {
        URCL_LOG_INFO("Robot requested program");
        sendProgram();
      }
      server_.disconnectClient();
    }
  }
```

fmauch, 2021-03-13:
Thanks for this report.

A better solution might be to actually refactor the socket implementation and get to a more event-driven system here. Every time I look at the socket implementation I find it quite confusing.

I would separate the client and server implementation to remove one layer of complexion. I'll hack together a proof of concept this weekend.

fmauch, 2021-03-14:
@urmahp Do you think that #46 could be a solution? If so, feel free to comment directly inside the PR.

closed 2021-04-10 by fmauch