UniversalRobots/Universal_Robots_Client_Library #58: Protective Stop after setting Speed slider to 0.0

MGBla, 2021-04-28:
After setting the the SpeedSlider to 0.0 using ``` ur_driver_->getRTDEWriter().sendSpeedSlider(0.0) ``` the next similar call with another fraction value puts the robot into protective stop with the following error message:
[image]

fmauch, 2021-05-01:
I guess, you are referring to the ROS2 driver? It would be interesting, whether this can be reproduced using only the RTDE interface, without any driver-related script code interfering.

MGBla, 2021-05-01:
No, the same behavior is (more or less) present in the ROS1 Driver. Although there the speed slider is simply restricted to values between [0.01 , 1.0]. I'm actually not sure if it is even a problem of the client library or if the issue lies below, since you cannot set a fraction smaller than 2% in the teach panel itself.

fmauch, 2021-05-01:
Well, both drivers run the same script code...

fmauch, 2021-05-07:
When setting it to 0.0 without anything else, nothing happens (at least in simulation).

I used the [example RTDE client](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/examples/rtde_client.cpp) and replaced the `sendSpeedSlider()` line with

```c++
    if (!my_client.getWriter().sendSpeedSlider(0.0))
```

While executing a motion on the robot TP (just a simple MoveJ with two waypoints), the robot stops as soon as I start the RTDE client. No protective stop. If I stop the RTDE client and increase the speed slider on the TP again, the robot continues moving.

I am unsure whether this behavior is URSim specific and a real robot might behave differently, though.

urmahp, 2021-05-07:
I ran into a similar issue, when working with Isaac driver when pausing the program, the protective stop only occurred on a real robot and not in simulation. The issue was related to the handle around pausing the program. If we send commands to the robot while the program is paused and resuming the protective stop is triggered, when resuming the program. Changing the Isaac driver to only send commands when the program is playing or pausing fixed the issue. In the ROS driver commands are also only sent when the program is playing or pausing. 

See how it is handled in the [ROS_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface.cpp#L537)

Setting the speed slider at 0 is in some way the same as pausing the program, when setting the speed slider to 0, we keep sending commands to the robot, so I think that is the reason that the protective stop is triggered, when changing the speed slider away from 0 again.

fmauch, 2021-05-07:
> Setting the speed slider at 0 is in some way the same as pausing the program, when setting the speed slider to 0, we keep sending commands to the robot, so I think that is the reason that the protective stop is triggered, when changing the speed slider away from 0 again.

As I understood it, it's the other way around: Pausing the program internally sets speed_scaling to 0. So yes, the behavior should be similar and yes, this is why we implemented the driver that way. @MGBla This should then be reproducible if pausing the program on the TP and playing it again. Do you handle this inside the ROS2 driver already?

MGBla, 2021-05-10:
Using the RTDE client example to set the speed slider to 0.0 and another value is working on the real hardware. However as @urmahp mentioned there a no other commands send to the robot in between. The pausing is handled in the [ROS2](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/53ee6b04035a1488bc3b5c5dbd8b5802e74c1aea/ur_robot_driver/src/hardware_interface.cpp#L482) similar to the ROS1 version and does not trigger the protective stop.

fmauch, 2022-09-27:
I think, this can be closed by now.

closed 2022-09-27 by fmauch