UniversalRobots/Universal_Robots_Client_Library #59: Pipeline producer overflowed

bblumberg, 2021-05-03:
Hi
As an experiment, I tried running the beta UR ROS2 driver on an Nvidia Xavier AGX (ARM processor), both in a docker container as well as standalone. In both cases, I get a runtime warning that "no realtime capabilities found", and then once the external node on the robot starts up, I get an alternating " Pipeline producer overflowed!" from line 412 of Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h followed by an attempt to reconnect, and then the overflow and so on.

Is this the expected behavior when not running with a real time kernel? In the absence of installing a realtime kernel is there a setting that can be set? Or does this not have anything to do with the realtime performance.

I assume this is a client library issue and not associated with the ROS2 driver itself, although I did note that it seemed like jointstates and io states were being published. The JointStateTrajectory controller was publishing commands, but the robot wasn't moving.

This was with the master branch of the client library

Thanks

bb

fmauch, 2021-05-15:
This is indeed a problem with the ROS2 driver.

The client library stores packages received from the robot inside a queue which has to be emptied (consumed) by a second party. If the queue fills up without being emptied, the mentioned warning is printed.

It is the driver's job to empty the queue, which should happen [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/5f49e74741c809a2568ea839326cf4026b1c847f/ur_robot_driver/src/hardware_interface.cpp#L405). I'll close this, as it looks like a ROS2 driver issue. Please open an issue inside the ROS2 driver repo linking to this issue. If any news concerning the library come up hunting this down, feel free to comment or reopen this one.

closed 2021-05-15 by fmauch

bblumberg, 2021-05-17:
Thanks. Posted on the ROS2 driver page as you suggested.

bb

fmauch, 2021-05-18:
For reference: The ROS2 driver issue is [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/113)