UniversalRobots/Universal_Robots_Client_Library #86: Reduce Delay Time of ServoJ Command ( lookahead_time ?)

yottasyno, 2021-08-11:
**Our Setup:**
We are communicating with the UR control box via TCP/IP using the RTDE protocol, sending commands from a linux system with patched real-time kernel to our UR10e robot. We want to control the robot in joint space and need very fast response to follow our desired joint angles that are sent from the linux pc. On the linux pc a loop with 2ms cycle time is running and sending new desired joint angles to the robot controler. On the robot side we are using the servoj(…) command to track the desired joint angles.

**Our Problem:**
We are saving the desired (sent to the robot) and actual joint angles (received from robot) in the same cycle as vectors on the linux system. After performing a cross-correlation of the two vectors, we see delays of ~17 time steps (with a cycle time of 2ms this corresponds to 34ms).
To test our communication we only sent joint angles to the controler and back to the linux system. (We removed the servoj() command and replaced it with a sync()). Now the delays shown by the cross-correlation are 2 or 3 time steps (corresponding to 4ms or 6ms).

**Our Conclusion:**
The servoj(q_des, 0, 0, 0.002, 0.03, 1200) has a time delay according to lookahead_time = 0.03s. This is a serious problem for our application.

**Our Question:**
Is the time delay of 30ms coming from the servoj() command with lookahead_time=0.03?
Can we somehow reduce this time delay?
Can we use another function than servoj() to control the joint angles? (maybe less safe but faster?)

Thanks for your help!!

stefanscherzinger, 2021-08-12:
Hi @yottasyno 

> Is the time delay of 30ms coming from the servoj() command with lookahead_time=0.03?

I'm not exactly sure but it appears highly related. Maybe @urrsk has some more insights on this?

> Can we somehow reduce this time delay?

At least the [UR script documentation](https://s3-eu-west-1.amazonaws.com/ur-support-site/68443/scriptManual.pdf) (p. 36)  suggests that the `lookahead time` be in range [0.03,0.2]. I cannot confirm right now if this is a requirement, and if so, what `UR script internal` mechanisms could exist to avoid this. The UR ROS driver, which is the reference implementation for ROS using this library, seems to [strictly adhere to those limits](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface.cpp#L148). I remember discussions with @fmauch on this topic during development of the UR ROS driver, and if I remember correctly, it was an *unsolvable* problem.

> Can we use another function than servoj() to control the joint angles? (maybe less safe but faster?)

If I oversee this correctly, `speedj()` ([p. 38](https://s3-eu-west-1.amazonaws.com/ur-support-site/68443/scriptManual.pdf)) does not seem to have this requirement. The reason might be that for internal position control, the robot controller needs to compute (as in estimate) meaningful motor torques to reach the desired joint positions in the given time, probably using some internal dynamics model and feed-forward torque control with PID correction on the remaining error. If you command velocities, this simplifies a little.

> We want to control the robot in joint space and need very fast response to follow our desired joint angles that are sent from the linux pc.

I'm not exactly sure about your high-level use case, but maybe using `speedj()` , i.e. wrapping your application around velocity control, is a suitable alternative? Note that instead of going through the RTDE interface, which might cause some additional delay, this library supports `speedj()` through the [ReverseInterface](https://github.com/UniversalRobots/Universal_Robots_Client_Library#reverseinterface).

gavanderhoorn, 2021-08-12:
It's been a few years since it was published -- so RTDE wasn't available yet -- but [Optimizing the Universal Robots ROS driver](https://orbit.dtu.dk/en/publications/optimizing-the-universal-robots-ros-driver) (written by @ThomasTimm seems like it could still be interesting.

yottasyno, 2021-08-13:
Hello everyone,

thanks a lot for the quick and detailed answers!

> If I oversee this correctly, speedj() (p. 38) does not seem to have this requirement. The reason might be that for internal position control, the robot controller needs to compute (as in estimate) [...]

I don't quite see why it would use the dynamic model for feed-forward control only for the position control or why this would take 30ms. If you have more insight I'd very interested to hear about that. Thanks a lot for the tip with the speedj() command, we will test the delay times with this function and post an update here.

> I'm not exactly sure about your high-level use case, but maybe using speedj() , i.e. wrapping your application around velocity control, is a suitable alternative? Note that instead of going through [...]

As far as we understood, the RTDE interface is the only protocol/interface that actually guarantees real-time properties. Can you tell us why the RTDE would be slower than the Reverse interface?

> Edit: I'm really guessing at the moment, but using spacenav with a 3DMaxon Spacemouse e.g. to do some teleoperation should neatly integrate with a velocity-based control approach.

Exactly, we did this for a small demo. Delay times are not so critical here, but future we want to sync 3 other motors with the UR and eventually have some force feedback.

urrsk, 2021-08-16:
If you want a fast reaction time, use the ```speedj()``` approach.
There are no speed or delay different between the RTDE and the Reverse interface. And both of them will suffer if the host can not keep up with the pace.

gavanderhoorn, 2021-08-16:
Which is indeed what the analysis by @ThomasTimm showed:

> As expected, the best performance is with speedj control [..]

fmauch, 2021-08-20:
Note: For ROS we also offer a velocity-based controller that uses speedj under the hood. I've created timing measurements back then, which show the same thing (using this library with the ReverseInterface): https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/1#issuecomment-549894865

You will always get a delay from the network connection between the linux machine and the controller.

Basically, you have the operations
 1. Send on Linux machine
 2. Receive on Controller
 3. start `speedj` command
 4. low level controllers execute command

You definitely lose a control cycle between 3 and 4. As you can see in my plot, we indeed reached a "reaction time" of one control cycle which is the minimum possible. If you look at the actual joint values, the delay will always be higher, as the controller also needs time to execute the commanded motion.

Sorry for writing even more text to this issue without really providing new information, but I felt it might be good to bring in our own actual measurements and reasoning, as well.