UniversalRobots/Universal_Robots_Client_Library #89: Unable to load script on the robot

hugols16, 2021-09-30:
The `ScriptSender` makes reference to the ExternalControl URCaps program that needs to be installed on the UR controller, and mentions that in order for the `ScriptSender` to work properly, the IP address must be set to the robot's IP address and the port must be set to the correct one (50002 in this case). Even with these settings properly set, the simple example `examples/full_driver.cpp` is able to get the full state of the robot but is unable to make J6 move, and the program state handler shows that the program isn't running. Are there missing steps in the documentation?

fmauch, 2021-10-04:
@hugols16 did you figure this out? If there is anything missing or unclear, please let us know so we can improve the documentation.

hugols16, 2021-10-04:
@fmauch I closed this issue because I thought I had figured out the issue at first, but it seems that it's not resolved after all. Here's a summary of what has been done to try and get the library to work with my UR5.
1. Install the External Control URCap on the UR5. I've then set up the IP address to be that of my computer (on the same network) and set the URCap port as 50002 (matching the default ScriptSender port from this library).
2. This library is running via docker so I've made sure to expose the port.
3. When I exec in docker, I'm able to not only ping the robot, showing that the robot IP address has been configured properly, but I'm also able to run a very simple RTDE program that controls the arm, again showing that communication can be established.
4. When not running the `full_driver` example or when exposing a different port, the program running the ExternalControl URCap on the UR5 will output that it can't establish a connection, thus stopping me from running said program. But when 50002 is exposed and when the `full_driver` example is running, the program running the ExternalControl URCap on the UR5 does not output any errors and stays running/playing.
5. With all that being said, the `full_driver` example outputs that the program isn't running. I've tried turning headless mode on and off, and that didn't do much to help. I've looked at the README countless times, and nothing else is said about a step I may be missing. I've thought of potential networking issue but then again, it seems that I'm able to establish a proper communication with the URCap. Maybe I'm missing a step when creating the program on the UR5 but I'm not quite sure. Right now the only command in the UR5 program is the ExternalControl command.

The `full_driver` example also outputs the following debug logs:
```DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 24 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 23 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 25 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/primary/primary_parser.h 139: Invalid robot package type recieved: 5 
```

However those are outputted after the example outputs that the program isn't running, and I'm unsure if it's related. Below is the full output from the full driver example:

```
DEBUG /opt/Universal_Robots_Client_Library/src/ur/ur_driver.cpp 63: Initializing urdriver 
DEBUG /opt/Universal_Robots_Client_Library/src/ur/ur_driver.cpp 64: Initializing RTDE client 
[... cut ...]  tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information] for details. 
DEBUG /opt/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h 345: Starting up producer 
WARN /opt/Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h 396: No realtime capabilities found. Consider using a realtime system for better performance 
```

hugols16, 2021-10-04:
Also to add to this investigation, I've made sure to expose all ports that were listed in the logs, i.e. 30001, 30002, 30004, 50001, 50002 and 50003. When the example is running and the program isn't running on the UR, but then I press play, I get the following logs:
```
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 209: Activity on self-pipe 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 233: Activity on FD 5 
DEBUG /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 46: New client connected at FD 15. 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 209: Activity on self-pipe 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 233: Activity on FD 15 
INFO /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 58: Robot requested program 
INFO /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 71: Sent program to robot 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 209: Activity on self-pipe 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 233: Activity on FD 15 
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 249: 15 disconnected. 
DEBUG /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 51: Client at FD 15 disconnected. 
```

It seems that the example is able to recognize that the URCap client connected and did send the robot program over, but then again no movement. The disconnected log is also a bit concerning.

fmauch, 2021-10-05:
The disconnection is not really a problem, as the script has been sent to the robot.

As you have already stated, the machine / container running the library / the example needs to expose 50001-50003 by default (reverse interface, script sender and trajectory interface).

As you are able to create a program on the robot containing an external_control node and it keeps running, it sounds (on a first glimpse) that everything is ok. The robot seems to connect to the script_sender, gets the script code and starts executing it.

Inside the code, the script [connects to the reverse interface](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/90338a76d65d86240710c8f976fd40296a49cd84/resources/external_control.urscript#L221) where it should receive joint commands frequently. If it doesn't it exits.

One problem that might be misleading is that by default a new program on the TP loops forever. So if the connection to the reverse interface fails, the program will stop, start again, try to connect to the reverse interface, stop, restart, ...

You should, however, see that in the logs on the controller. Could you check whether there is anything unusual going on in the logs?

cindyysi, 2022-07-26:
@fmauch I encounter the same issue that I am not able to play the program by remote control.  The external control URCap was downloaded from[ https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/externalcontrol-1.0.5.urcap](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/externalcontrol-1.0.5.urcap) and I follow the steps in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/install_urcap_e_series.md to use this URCap by adding a program to load this external control node.  The script sent to the robot is download from https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/resources/external_control.urscript and I change `SERVER_IP_REPLACE` in https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/resources/external_control.urscript#L221-L222 to be hardcoded as my laptop's IP (`192.168.1.55`) to assure it does not pick up the docker's IP.

I am using the dashboard to send commands to load and run the program.
1. I use `load` to load the `.urp` program.  I see the below log as the response
```Loading program: /programs/external_control.urp, /programs/default.installation```

2. I confirm that the program is loaded completely by `get loaded program` and the response is as below
```Loaded program: /programs/external_control.urp```

3. Then I use `play` to run the program.  The below log is printed
```
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 209: Activity on self-pipe
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 233: Activity on FD 11
DEBUG /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 46: New client connected at FD 22.
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 209: Activity on self-pipe
DEBUG /opt/Universal_Robots_Client_Library/src/comm/tcp_server.cpp 233: Activity on FD 22
INFO /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 58: Robot requested program
INFO /opt/Universal_Robots_Client_Library/src/control/script_sender.cpp 71: Sent program to robot 
```

I check that the reverse_ip is set correctly (same as my laptop's, `192.168.1.55`) and the robot's IP is `192.168.1.102`.  The port forwarding is as expected
```$ docker container ls
CONTAINER ID   IMAGE          COMMAND                  CREATED              STATUS              PORTS                                                                                                                          NAMES
c20a7c483ad7   c8d9a50ee004   "/tmp/platform/cmake…"   About a minute ago   Up About a minute   0.0.0.0:9000->9000/tcp, 0.0.0.0:29999->29999/tcp, 0.0.0.0:30001-30004->30001-30004/tcp, 0.0.0.0:50001-50004->50001-50004/tcp   brave_dirac 
```
and the firewall is inactive
```
$ sudo ufw status verbose
Status: inactive
```
However, it seems like I still have problem establishing connection between the robot and my laptop
```
$ ss -atnp | grep 5000
LISTEN     0       4096                0.0.0.0:50001              0.0.0.0:*                                                                                     
LISTEN     0       4096                0.0.0.0:50002              0.0.0.0:*                                                                                     
LISTEN     0       4096                0.0.0.0:50003              0.0.0.0:*                                                                                     
LISTEN     0       4096                0.0.0.0:50004              0.0.0.0:*     
```

Any idea/suggestion is appreciated, thanks!

fmauch, 2022-09-27:
@cindyysi which program did you run? Does the plain rtde_client example work correctly?

fmauch, 2024-06-12:
Closing this due to inactivity. Feel free to re-open / comment.

closed 2024-06-12 by fmauch