UniversalRobots/Universal_Robots_Client_Library #91: Reading data from module internals of URScript #501

VishnuDevA7, 2021-12-31:
I am trying to read data from one of the variables declared inside the URScript file. For instance, to check for safety aspects, I declare a variable inside *.urscript file that uses the module internals function 'global joint_torques = get_joint_torques()' as described in the Script Manual of Universal Robots. While the URScript file has access to this data, can this data be accessed by a high-level script where I implement my fancy controller? If so, how?

urrsk, 2022-01-05:
@VishnuDevA7 currently, the actual torque on the joints are not publish by the controller.
One work around is to add URScript code that saves the return values from get_joint_torques() to six double_register, thereby you are able to receive them through RTDE.
See http://help.universal-robots.com/OTHER/ScriptwebG5/Content/prod-scriptmanual/G5/write_output_float_register.htm?Highlight=write_output_float_register

fmauch, 2022-09-27:
Closing due to inactivity.

closed 2022-09-27 by fmauch