UniversalRobots/Universal_Robots_Client_Library #94: startup before booted robot does not work

fmauch, 2022-02-08:
Starting up the client library interfaces before the robot is booted leads to multiple problems. This has been reported in the ROS driver, e.g. in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/272

Basically, we have to distinguish two cases:

If the robot is not connected / the IP address is configured wrong, opening a connection to the remote socket fails immediately [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/8f20b42205f645bba406d5cf3b1163ea0a42fbc9/src/comm/tcp_socket.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L87) leading to [this exception](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/8f20b42205f645bba406d5cf3b1163ea0a42fbc9/include/ur_client_library/comm/producer.h?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L71). Mitigating this would require having a retry mechanism either in the TCPSocket or inside the producer which would effectively block the startup process until a connection can be established.

When the robot is actually connected, but not fully booted, the connection described above works. The primary interface connection seems to be coming up and seems to be usable afterwards (did a quick test using the primary_interface example and a URSim docker container). The RTDE-Interface, however, seems to have more problems. As far as I know the RTDE-Interface on the robot side is restarted at some point. So, if we are too early, the restart either happens in the middle of the handshake (I had a test run, where the client was waiting for confirming the configured communication frequency) or it the handshake is performed, but then the client is waiting for new data which it will never get, as the restarted RTDE server never made a proper handshake with the driver. This would effectively require the RTDEClient to be able to go back to an uninitialized state either when no new data is received or when the handshaking process is not behaving as expected (timeouting at some point). In that case we want to restart the handshake automatically (with accepting that we might have to wait for an answer for an undefined time with the first handshake step).

I hope that helps someone implementing this, maybe @urmahp you want to have a go at this?

urmahp, 2022-02-11:
I will be happy to try and implement this. However, I might not have time to look into this before Friday next week.

fmauch, 2022-04-12:
fixed in #98

closed 2022-04-12 by fmauch