UniversalRobots/Universal_Robots_Client_Library #97: Error "ur_robot_driver" with "catkin_make_isolated" ("error: ‘urcl’ has not been declared"....)

totototo96, 2022-03-30:
Hi Everyone,
I've the problem with `ur_robot_driver` when I do `catkin_make_isolated` for building all the drivers of `Universal_Robots_ROS_Driver `. 
System Version: 
- Ubuntu 16.04
- Ros: Kinetic

This is the error occurred: 

>==> make -j4 -l4 in '/root/catkin_ws/build_isolated/ur_robot_driver'
Scanning dependencies of target cartesian_control_msgs_generate_messages_cpp
Scanning dependencies of target cartesian_control_msgs_generate_messages_nodejs
Scanning dependencies of target trajectory_msgs_generate_messages_nodejs
Scanning dependencies of target trajectory_msgs_generate_messages_cpp
[  0%] Built target cartesian_control_msgs_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_nodejs
[  0%] Built target cartesian_control_msgs_generate_messages_cpp
[  0%] Built target trajectory_msgs_generate_messages_cpp
Scanning dependencies of target control_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_cpp
Scanning dependencies of target roscpp_generate_messages_eus
[  0%] Built target control_msgs_generate_messages_lisp
Scanning dependencies of target actionlib_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
Scanning dependencies of target actionlib_msgs_generate_messages_eus
Scanning dependencies of target control_toolbox_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_lisp
Scanning dependencies of target control_toolbox_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target control_toolbox_generate_messages_nodejs
Scanning dependencies of target pass_through_controllers_gencfg
Scanning dependencies of target rosgraph_msgs_generate_messages_eus
[  0%] Built target control_toolbox_generate_messages_py
[  0%] Built target pass_through_controllers_gencfg
Scanning dependencies of target std_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_eus
Scanning dependencies of target ur_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
Scanning dependencies of target actionlib_msgs_generate_messages_nodejs
Scanning dependencies of target tf2_msgs_generate_messages_py
Scanning dependencies of target actionlib_generate_messages_cpp
[  0%] Built target ur_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target actionlib_generate_messages_cpp
Scanning dependencies of target rosgraph_msgs_generate_messages_py
Scanning dependencies of target tf_generate_messages_lisp
Scanning dependencies of target sensor_msgs_generate_messages_py
Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_lisp
Scanning dependencies of target roscpp_generate_messages_nodejs
Scanning dependencies of target trajectory_msgs_generate_messages_lisp
Scanning dependencies of target std_srvs_generate_messages_nodejs
Scanning dependencies of target sensor_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target trajectory_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_srvs_generate_messages_nodejs
Scanning dependencies of target control_msgs_generate_messages_py
Scanning dependencies of target actionlib_generate_messages_nodejs
Scanning dependencies of target actionlib_msgs_generate_messages_cpp
[  0%] Built target control_msgs_generate_messages_py
Scanning dependencies of target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_cpp
Scanning dependencies of target std_srvs_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_lisp
[... 270 lines cut ...]
   ur_driver_->writeTrajectoryControlMessage(urcl::control::TrajectoryControlMessage::TRAJECTORY_START, point_number);
               ^
/root/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1258:17: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryPoint’
     ur_driver_->writeTrajectoryPoint(p, true, next_time - last_time);
                 ^
/root/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function ‘void ur_driver::HardwareInterface::cancelInterpolation()’:
/root/catkin_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1267:15: error: ‘class urcl::UrDriver’ has no member named ‘writeTrajectoryControlMessage’
   ur_driver_->writeTrajectoryControlMessage(urcl::control::TrajectoryControlMessage::TRAJECTORY_CANCEL);
               ^
CMakeFiles/ur_robot_driver_plugin.dir/build.make:86: recipe for target 'CMakeFiles/ur_robot_driver_plugin.dir/src/hardware_interface.cpp.o' failed
make[2]: *** [CMakeFiles/ur_robot_driver_plugin.dir/src/hardware_interface.cpp.o] Error 1
CMakeFiles/Makefile2:321: recipe for target 'CMakeFiles/robot_state_helper.dir/all' failed
make[1]: *** [CMakeFiles/robot_state_helper.dir/all] Error 2
CMakeFiles/Makefile2:619: recipe for target 'CMakeFiles/ur_robot_driver_plugin.dir/all' failed
make[1]: *** [CMakeFiles/ur_robot_driver_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'ur_robot_driver': 
  Command '['/root/catkin_ws/devel_isolated/ur_kinematics/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2
Reproduce this error by running:
==> cd /root/catkin_ws/build_isolated/ur_robot_driver && /root/catkin_ws/devel_isolated/ur_kinematics/env.sh make -j4 -l4
Command failed, exiting.

Can you help me guys?
Thanks a lot for your time!

wj5449, 2022-08-26:
@fmauch 
Hi 
i want to control the robot with c++ code , i wonder which project should i reference for.  
or  any other suggestions?

fmauch, 2022-09-19:
@wj5449 I don't see why this would be related to the OP. Hijacking and @-mentioning maintainers is not a very nice thing to do inside an open-source community.

That question seems to be better asked on ROS answers, where there will be a broader audience and the question is not restricted to UR robots. As a very short answer: You basically want to have an action client on the scaled_joint_trajectory_controller. Filling in goals for that can be rather tedious which is why most people would use MoveIt! for that. With MoveIt! you would have a MoveGroupInterface in C++ which you use to talk to MoveIt!. There are numerous tutorials about writing a MoveGroupInterface out there.