UniversalRobots/Universal_Robots_Client_Library #122: Error while compiling this workspace

marioesteban11, 2022-10-07:
Hi,

This error shows up when I`m trying to compile this workspace following the README Instructions.

This is the error that shows up when im trying to compile this workspace using; 

# colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

ERROR: 

## --- stderr: ur_calibration                                                                                                                                                                       
/usr/bin/ld: /home/marioest/workspace/ros_ur_driver/install/ur_robot_driver/lib/libur_robot_driver_plugin.so: undefined reference to `urcl::UrDriver::UrDriver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (bool)>, bool, std::unique_ptr<urcl::ToolCommSetup, std::default_delete<urcl::ToolCommSetup> >, unsigned int, unsigned int, int, double, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
collect2: error: ld returned 1 exit status

I`m using:

Ubuntu 22.04
ROS2 Humble

The objetive is do a pick and place planning using UR and ROS2.

Thanks in advance

fmauch, 2022-10-10:
Did you operate on a freshly cloned workspace? As you can imagine - that's not supposed to happen and it seems not to happen neither in our CI pipelines nor our dev machines.

That being said we are in the process of an API extension in the client library which we would expect to not cause any problems. If it does cause issues, we are very interested in understanding why.

Therefore, some more information would be helpful:

- Do you work on a freshly cloned workspace?
- Do you have the binary package `ros-humble-ur-client-library` installed? If yes, in which version?
- What were the exact commands from workspace setup until `colcon build`? (Relevant whether you have some other packages in your workspace due to a `vcs import` command for example.

If you are only interested in getting it running and you don't care about development of the driver, simply install the binary packages by running `sudo apt install ros-humble-ur`.

fmauch, 2022-10-10:
This seems to be a cross post with https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/500.

I've encountered a similar issue compiling the ROS1 driver while having the library installed as binary version and a newer library version as source package.

marioesteban11, 2022-10-10:
Hi fmauch. Finally i was abla to compile it just rebuilding the whole workspace. I deleted build, log and install forlder and after that  I compile it again and it is working correctly right now.

Thanks for the comment :)

closed 2022-10-10 by marioesteban11