UniversalRobots/Universal_Robots_Client_Library #142: Uninformative error when script file does not exist

carebare47, 2023-03-06:
Hi all,

More of a suggestion than an issue. Due to a mismatch in versions between our configs and the `Universal_Robots_ROS_Driver`, we ended up passing a filename that didn't exist as the `script_file` name [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/src/hardware_interface.cpp#L126-L130) which ends up [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/2e69bc165da0ad22fe9a1a3417e7eaa4b082ebac/src/ur/ur_driver.cpp#L90).

There is already a check that the parameter exists in the ROS driver, but if the parameter exists and points to an invalid file, the invalid file path gets passed down into this client library and throws weird errors:
```
basic_string::replace: __pos (which is 18446744073709551615) > this->size() (which is 0)
[ERROR](/sr_hand_robot): [1678106531.818786780] Initializing robot HW 'ra_sr_ur_robot_hw' failed
ERROR /tmp/binarydeb/ros-noetic-ur-client-library-1.3.1/src/rtde/rtde_client.cpp 537: Could not get response to RTDE communication pause request from robot 
```

Is there any reason not to add a check [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/2e69bc165da0ad22fe9a1a3417e7eaa4b082ebac/src/ur/ur_driver.cpp#L332-L338) that the file exists and is valid?

RobertWilbrandt, 2023-03-07:
I don't see any reason to not add that, it seems like a straightforward improvement. I'd be happy to merge a pull request implementing this, otherwise i'll try to add this myself in the next weeks.

jeferal, 2023-06-07:
Thank you. Are you working on adding this check at the moment?

PR #173 (2023-08-31): Throws exception if the URScript file doesn't exists
This will fix #142
commit cb9c7f32 (2023-08-31): Throws exception if the URScript file doesn't exists (#173)

This will fix #142
closed 2023-08-31 by fmauch