UniversalRobots/Universal_Robots_Client_Library #156: reconnect the robot after connection dropped down.

ZhengHaichao, 2023-07-07:
Hi Team,

A OEM customer is using this library in their product, a disconnection issue affects their development, so a method of reconnecting the robot is needed.

They are using the method in example/full_driver.cpp to create the connection, but when the connection dropped down, how to reconnect the robot in another thread? without closing their main program.

Thanks.

fmauch, 2023-07-07:
If the connection is dropped, this is usually due to a non-realtime communication between the control pc and the robot controller. Our current suggestion is to use a real-time-patched kernel or at least a lowlatency kernel on the control pc side and a dedicated network connection (direct cable between the control pc and the robot controller, no switch).

We plan to release a version with options for a less restrictive communication requirements, but for now a stable and steady communication is required.

Depending on the use case you can also drop the requirements manually, by calling `setKeepaliveCount(5)` on the driver object, as recently updated in the example:

https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/57c1a6a4bca78c36ac652b6d3bf43dd56fb38862/examples/full_driver.cpp#L113-L114

However, this has the side effect that the robot will extrapolate on its own, when no updates from the pc side come in. So, when relying on setpoint streaming, this is not recommended.

Kellen2021, 2023-07-17:
> If the connection is dropped, this is usually due to a non-realtime communication between the control pc and the robot controller. Our current suggestion is to use a real-time-patched kernel or at [...]

Hello, I have used this library in my software, but I am experiencing a remote connection disconnect issue. The specific error is  as follows:
`Failed to read from stream, reconnecting in 32 seconds...`.
Is it possible to achieve a reconnection without restarting the software using this library? I noticed that the DashboardClient class has connect and disconnect methods. Can reconnection for communication on other ports be achieved using a similar approach?

fmauch, 2023-07-17:
Automatic re-connection is partly built-in and is on our schedule, but not at highest priority at the moment.

zhoubo2018, 2024-12-16:
Hello, sir,
I also have some problem about network disconnection. I used the API of ur_rtde_master to connect to robot.
The description of my problem:
When robot's network has disconnected, the program would terminate/crash/shutdown. The break point of program is at "RTDEReceiveInterface rtde_recceive(ip_address)" or "RTDEControlInterface rtde_control(ip_address)", here the "ip_address" is a string member, I see an interface named "reconnect()" and "isconnect()", but when I use that, it could not work properly.
Because I can't see the detail defination of rtde API, so could you tell me how to write the "reconnect()" function rightly to avoid the program crash when I use rtde API ? 
Thanks.

urfeex, 2024-12-18:
> Hello, sir, I also have some problem about network disconnection. I used the API of ur_rtde_master to connect to robot. The description of my problem: When robot's network has disconnected, the [...]

I'm not sure whether I understand things correctly. Are you using ur_rtde or the ur_client_library. If the first  is the case, then I cannot really give any advice, as I am not familiar with ur_rtde. In the latter case: What exactly are you trying to achieve? What components are you running?