UniversalRobots/Universal_Robots_Client_Library #191: [start_ursim] Script does not check ursim version

RobertWilbrandt, 2023-12-06:
`start_ursim` does not check if the used ursim version is up-to-date. This can lead to the following problem:
1. Having used the script earlier, docker pulled a version that was current at the time of use.
2. The ur20 is released and the driver now passes this robot model to ursim.
3. The old version does not know that version string and defaults to a ur5e.

This leads to no visible errors and unexpected results.

gavanderhoorn, 2023-12-06:
I believe this is a related issue: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/842.

VinDp, 2024-05-29:
This issue is solved by [#203](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/203).

PR #203 (2024-05-29): [start_ursim] Always check ursim version for compatibility
Before, the version was only checked when explicitly specified. This could lead to errors if there was a local image with tag "latest" which was actually an old version.

This closes #191 

I reproduced #191 locally by explicitly pulling an old image and tagging it "latest". With this PR applied things look like this:

```bash
dev@localhost:~$ docker image ls | grep e-series
universalrobots/ursim_e-series                                                 5.12.2                 21974ae9b4ac   20 months ago   1.15GB
universalrobots/ursim_e [... cut]
closed 2024-05-29 by VinDp