UniversalRobots/Universal_Robots_Client_Library #209: Compilation failure with clang

remi-siffert-ocado, 2024-07-17:
### Affected Client Library version(s)

1.3.7

### What combination of platform is the Client Library running on.

Linux

### How did you install the Client Library

Build the library from source

### Which robot platform is the library connected to.

URSim in docker

### Robot SW / URSim version(s)

5.16.0

### How are you using the Client Library

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The library cannot be compiled with `clang` because of an error in the full driver example.

## Issue details
The [following line](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/613b2f9f81ed888c2fac9463963d6dc9d07ba762/examples/full_driver.cpp#L163 ) in `full_driver.cpp` fails to compile with: 

`universal_robots_client_library/examples/full_driver.cpp:163:39: error: cannot pass object of non-trivial type 'std::string' (aka 'basic_string<char>') through variadic function; call will abort at runtime [-Wnon-pod-varargs]
      URCL_LOG_DEBUG("data_pkg:\n%s", data_pkg->toString());`

## Steps to Reproduce
- Clone the repository in version 1.3.7.
- In the cloned repository: `cmake -DCMAKE_CXX_COMPILER=clang++-12 -DCMAKE_C_COMPILER=clang-12 .` and `make`.

## Expected Behavior
The library should compile successfully with `gcc` and `clang`.

## Actual Behavior
The library compiles successfully with `gcc` but fails to compile with `clang`.

# Workaround Suggestion
Replacing `URCL_LOG_DEBUG("data_pkg:\n%s", data_pkg->toString());` with `URCL_LOG_DEBUG("data_pkg:\n%s", data_pkg->toString().c_str());` does the trick.
Happy to open a PR for this.

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-07-18:
This seems fair, thank you!

remi-siffert-ocado, 2024-07-23:
PR https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/210 is open for review.

closed 2024-07-26 by fmauch