UniversalRobots/Universal_Robots_Client_Library #240: Add documentation of how to make it work with Polyscope 5.19 and PolyScopeX, due to the new security settings.

urrsk, 2025-01-07:
### Feature summary

From PolyScope 5.19 and newer, including PolyScopeX, it is needed to Enable the Dashboard, Primary, Secondary, and Realtime Data Exchange Interfaces through PolyScope's Security Service Settings in order to make the make the Client Library communicate with the robot.

[image]

### Tasks
*To complete this issue involves*
- [ ] Make documentation
- [ ] Make a note as well in the ROS drivers
- [ ] Optional: Make note as well in the [Universal_Robots_ExternalControl_URCap](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap) and [Universal_Robots_ExternalControl_URCapX](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCapX)

urrsk, 2025-01-07:
Also mentioned on https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1196

urfeex, 2025-01-08:
We currently don't even have the robot setup part in the client library's documentation at the moment. Maybe we should move it from the ROS drivers to the library and refer to it from the driver documentation?

urrsk, 2025-01-08:
> We currently don't even have the robot setup part in the client library's documentation at the moment. Maybe we should move it from the ROS drivers to the library and refer to it from the driver documentation?

Good idea

closed 2025-01-27 by urfeex