UniversalRobots/Universal_Robots_Client_Library #278: There is an error in the external_control.script file in version 1.6.0

hestrro, 2025-02-26:
### Affected Client Library version(s)

1.6.0

### What combination of platform is the Client Library running on.

Linux in a virtual machine

### How did you install the Client Library

From ROS binary packets

### Which robot platform is the library connected to.

Real robot

### Robot SW / URSim version(s)

ur5e

### How are you using the Client Library

Headless without using the teach pendant

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*
There is an error in the external_control.script file in version 1.6.0
He explained in the script that the speed and speed parameters of the trajectory points were written backwards, so when calling the function, v needs to be written in the front and a in the back.

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-02-26:
Could you please be a bit more specific? Which call / line are you referring to?

hestrro, 2025-02-26:
[image]
Two weeks ago, when I tested the real robot arm of ur5e, the positions of the v and a parameters were reversed.
by textmsg打印

urfeex, 2025-02-26:
Ah, you are referring to the order in the byte stream?

Yes, that looks correct. Thanks for pointing that out.

hestrro, 2025-02-26:
I see that your clent library has been updated very quickly recently. Will it be troublesome if I want to change to a new version later?

urfeex, 2025-02-26:
> I see that your clent library has been updated very quickly recently. Will it be troublesome if I want to change to a new version later?

As long as the major version isn't changing, all introduced changes should be backwards-compatible. Please see the [changelog](https://github.com/UniversalRobots/Universal_Robots_Client_Library/releases) for details on the releases.

PR #279 (2025-02-26): Fix buffer order of acceleration and velocity
This was read in the reverse order from the script. The order used in the buffer follows the strategy that for splines it is possible to use velocities while not providing accelerations, hence they are in the buffer earlier.

---

Fixes #278 

---

We could also potentially refactor the code to change the order inside the buffer, but that would require refactoring the whole spline interpolation pipeline, as well.
closed 2025-02-26 by urfeex