UniversalRobots/Universal_Robots_Client_Library #312: It is suggested to add auxiliary methods about forward kinematics and inverse kinematics of the robot arm to the ur client library

hestrro, 2025-04-22:
### Feature summary

*Short introduction to the feature and why it should be implemented*

*Followed by a longer description*

### Related issues
*Links to related issues internal or external to this repo*

### Tasks
*To complete this issue involves*
- [ ] Implement the feature
- [ ] Make documentation
- [ ] Make Unit test
- [ ] Make example
- [ ] Test on real hardware

hestrro, 2025-04-23:
I'd like to convert Cartesian points and joint angles to and from each other, but I've just seen you implement this in your script and in your client library as a pose enumeration

hestrro, 2025-04-23:
g_my_robot->getUrDriver()->writeJointCommand(tcp_target, comm::ControlMode::MODE_POSE,
                                               RobotReceiveTimeout::millisec(200));
I can't control its velocity and acceleration through the writeJointCommand function. I need to convert the Cartesian coordinates to joint angles, so I hope you can give me a conversion function for inverse kinematics, or is there one I haven't found?

hestrro, 2025-04-24:
For example, when my robot arm moves with the car, I use some methods to find the gripping coordinates of the robot arm in the new position (Cartesian coordinates), and then I need to use movej to move directly above the new gripping coordinates, at this time I need to convert the Cartesian coordinates to the joint angle (because this saves time and the robot arm has to move significantly)，Regardless of whether you can add an auxiliary method, please reply to me (I'm in a hurry to use it here), thank you very much

urfeex, 2025-04-25:
Such things currently don't exist, no. If I understand you correctly, having the possibility to send Cartesian coordinates to moveJ would fit your usecase. While this is possible on a urscript level, this is currently not possible with the C++ API. As a workaround, you can also send URScript functions directly using the [`sendScript`](https://docs.ros.org/en/rolling/p/ur_client_library/generated/classurcl_1_1UrDriver.html#_CPPv4N4urcl8UrDriver10sendScriptERKNSt6stringE) function of the `UrDriver`. This is a fire-and-forget interface, however, meaning that you won't get any direct feedback on executing your motion.

hestrro, 2025-04-25:
The corresponding function in urscript is get_inverse_kin(x, qnear, maxPositionError=1e-10,
maxOrientationError=1e-10, tcp=’active_tcp’). Is it impossible to implement such a function in C++?

urfeex, 2025-04-25:
The technical difficulty would be to send the response back to the client_library application.

As a standalone feature you could add a function calling that to the [ScriptCommandInterface](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/control/script_command_interface.cpp). You would have to extend the [callback](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/control/script_command_interface.cpp#L252) to handle a different message that could contain the result from the inverse kinematic calculation. At this point it would make sense to put a bit more thoughts into the communication protocol there, however. For a works-for-you implementation you could make that a bit simpler and just take the number of bytes as a decision.

hestrro, 2025-04-29:
Maybe i can add one more motiontype in writemotionprimitive, do movej by cartesian coordinates, and then make corresponding changes in external_control.urscript

urfeex, 2025-04-29:
If you would like to do that, feel free to create a PR. We will potentially provide this feature later this year.