UniversalRobots/Universal_Robots_Client_Library #313: writeJointCommand'problem

hestrro, 2025-04-23:
### Affected Client Library version(s)

1.9.0

### What combination of platform is the Client Library running on.

Linux

### How did you install the Client Library

From ROS binary packets

### Which robot platform is the library connected to.

URSim in linux

### Robot SW / URSim version(s)

URSim_VIRTUAL-5.12.6.1102099.vmdk.lck

### How are you using the Client Library

Headless without using the teach pendant

### Issue details

[image]

# Summary
*Short introduction to the issue and how it impact you and why*

## Issue details

*Detailed description help us understand the problem. Code are welcome!*
I want to convert Cartesian coordinates into joint angles, but I cannot control the speed and acceleration, resulting in the above picture.

// -- BEGIN LICENSE BLOCK ----------------------------------------------
// Copyright 2023 Universal Robots A/S
//
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions are met:
//
//    * Redistributions of source code must retain the above copyright
//      notice, this list of conditions and the following disclaimer.
//
//    * Redistributions in binary form must reproduce the above copyright
//      notice, this list of conditions and the following disclaimer in the
//      documentation and/or other materials provided with the distribution.
//
//    * Neither the name of the {copyright_holder} nor the names of its
//      contributors may be used to endorse or promote products derived from
//      this software without specific prior written permission.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
// AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
// IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
// ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
// LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
// CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
// SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
// INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
// CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
// ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
// POSSIBILITY OF SUCH DAMAGE.
// -- END LICENSE BLOCK ------------------------------------------------

#include <gtest/gtest.h>

#include <ur_client_library/ur/dashboard_client.h>
#include <ur_client_library/ur/ur_driver.h>
#include <ur_client_library/example_robot_wrapper.h>

using namespace urcl;

const std::string SCRIPT_FILE ="/home/hy/ros2_ws/src/sto_ur_controller/resources/external_control.urscript";
const std::string OUTPUT_RECIPE = "/home/hy/ros2_ws/src/sto_ur_controller/params/rtde_output_recipe.txt";
const std::string INPUT_RECIPE = "/home/hy/ros2_ws/src/sto_ur_controller/params/rtde_input_recipe.txt";
const std::string CALIBRATION_CHECKSUM = "calib_12788084448423163542";
std::string g_ROBOT_IP = "192.168.30.128";
bool g_HEADLESS = true;

std::unique_ptr<ExampleRobotWrapper> g_my_robot;

class UrDriverTest : public ::testing::Test
{
protected:
  static void SetUpTestSuite()
  {
    
    // Setup driver
    g_my_robot = std::make_unique<ExampleRobotWrapper>(g_ROBOT_IP, OUTPUT_RECIPE, INPUT_RECIPE, g_HEADLESS,
                                                       "external_control.urp", SCRIPT_FILE);

    g_my_robot->startRTDECommununication(true);
  }

  void SetUp()
  {
    if (!g_my_robot->isHealthy())
    {
      ASSERT_TRUE(g_my_robot->resendRobotProgram());
      ASSERT_TRUE(g_my_robot->waitForProgramRunning(500));
    }
  }

  void TearDown()
  {
    g_my_robot->getUrDriver()->stopControl();
    g_my_robot->waitForProgramNotRunning(1000);
  }
};

TEST_F(UrDriverTest, target_outside_limits_pose)
{
  // g_my_robot->stopConsumingRTDEData();
  // std::unique_ptr<rtde_interface::DataPackage> data_pkg;
  // g_my_robot->readDataPackage(data_pkg);

  // urcl::vector6d_t tcp_pose_before;
  // ASSERT_TRUE(data_pkg->getData("actual_TCP_pose", tcp_pose_before));

  // Create physically unfeasible target
  // urcl::vector6d_t tcp_target = tcp_pose_before;
  // tcp_target[2] += 0.3;

  // Send unfeasible targets to the robot
  // g_my_robot->readDataPackage(data_pkg);
  urcl::vector6d_t tcp_target={0.124,-0.263,0.184,0.004,-0.01,0.796};
  g_my_robot->getUrDriver()->writeJointCommand(tcp_target, comm::ControlMode::MODE_POSE,
                                               RobotReceiveTimeout::millisec(200));

  // // Ensure that the robot didn't move
  // g_my_robot->readDataPackage(data_pkg);
  // urcl::vector6d_t tcp_pose;
  // ASSERT_TRUE(data_pkg->getData("actual_TCP_pose", tcp_pose));
  // for (unsigned int i = 0; i < 6; ++i)
  // {
  //   EXPECT_FLOAT_EQ(tcp_pose_before[i], tcp_pose[i]);
  // }

  // Make sure the program is stopped
  g_my_robot->startConsumingRTDEData();
  g_my_robot->getUrDriver()->stopControl();
  g_my_robot->waitForProgramNotRunning(1000);
}

int main(int argc, char* argv[])
{
  ::testing::InitGoogleTest(&argc, argv);

  for (int i = 0; i < argc; i++)
  {
    if (std::string(argv[i]) == "--robot_ip" && i + 1 < argc)
    {
      g_ROBOT_IP = argv[i + 1];
      break;
    }
    if (std::string(argv[i]) == "--headless" && i + 1 < argc)
    {
      std::string headless = argv[i + 1];
      g_HEADLESS = headless == "true" || headless == "1" || headless == "True" || headless == "TRUE";
      break;
    }
  }

  return RUN_ALL_TESTS();
}

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell
Note: Google Test filter = UrDriverTest.target_outside_limits_pose
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from UrDriverTest
[1745386320.032326] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/primary/primary_client.cpp 67: Starting primary client pipeline 
[1745386320.086940] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/ur/dashboard_client.cpp 76: Connected: Universal Robots Dashboard Server
 
[1745386320.140250] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/example_robot_wrapper.cpp 201: Robot ready to start a program 
[1745386320.141304] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/helpers.cpp 94: SCHED_FIFO OK, priority 99 
[1745386320.353476] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/rtde/rtde_client.cpp 139: Negotiated RTDE protocol version to 2. 
[1745386320.354293] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/rtde/rtde_client.cpp 291: Setting up RTDE communication with frequency 500.000000 
Program running: false

[1745386321.379753] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/primary/primary_client.cpp 67: Starting primary client pipeline 
[1745386321.489053] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/control/reverse_interface.cpp 219: Robot connected to reverse interface. Ready to receive control commands. 
Program running: true

[1745386321.489586] INFO /home/hy/ros2_ws/src/Universal_Robots_Client_Library/src/helpers.cpp 94: SCHED_FIFO OK, priority 99 
[ RUN      ] UrDriverTest.target_outside_limits_pose
terminate called without an active exception
Aborted (core dumped)
```

### Accept Public visibility

- [x] I agree to make this context public

urrsk, 2025-04-25:
@hestrro without looking too much into it. I will conclude that your target have a too long distance to the current robot position that the robot can reach within 2ms. That is why it reports that it is exceeding the robot limit.
The ```writeJointCommand``` is an interface for streaming realtime joint commands. Therefore, you also need to start your motion interpolation from where the robots currently is and not just send the final target you what the robot to go to.

If you just what to send the final target of the waypoint, I will suggest you using the ```writeTrajectoryPoint``` function instead or the new ```writeMotionPrimitive``` function. See https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/include/ur_client_library/ur/ur_driver.h

closed 2025-04-25 by urrsk

urfeex, 2025-04-25:
Please also see our [InstructionExecutor example](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/examples/instruction_executor.html) on sending distant targets with a one-line-call.

hestrro, 2025-04-25:
I use writeJointCommand instead of writeTrajectoryPoint because I found that when the controlmode in externalcontrol.urscript is pose, the script function get_inverse_kin will be called. I want to perform an inverse kinematic conversion from Cartesian to joint angles.This is a very useful function. I hope you can consider adding it to the C++ interface.

get_inverse_kin(x, qnear, maxPositionError=1e-10,
maxOrientationError=1e-10, tcp=’active_tcp’)
Calculate the inverse kinematic transformation (tool space -> joint
space). If qnear is defined, the solution closest to qnear is returned.
Otherwise, the solution closest to the current joint positions is returned. If
no tcp is provided the currently active tcp of the controller will be used.
Throws a runtime exception if no solution was found.
Parameters
x: tool pose
qnear: list of joint positions (Optional)
maxPositionError: the maximum allowed position
error (Optional)
maxOrientationError: the maximum allowed orientation
error (Optional)
tcp: tcp offset pose (Optional)
Return Value
joint positions
Example command: get_inverse_kin(p[.1,.2,.2,0,3.14,0],
[0.,3.14,1.57,.785,0,0])
• Example Parameters:
– x = p[.1,.2,.2,0,3.14,0] → pose with position of x=100mm,
y=200mm, z=200mm and rotation vector of rx=0 deg., ry=180
deg, rz=0 deg.
– qnear = [0.,3.14,1.57,.785,0,0] → solution should be near to
joint angles of j0=0 deg, j1=180 deg, j2=90 deg, j3=45 deg,
j4=0 deg, j5=0 deg.
– maxPositionError is by default 1e-10 m
– maxOrientationError is by default 1e0

hestrro, 2025-04-29:
Could you tell me what situations this writejointcommond function is generally used in? Thank you

urfeex, 2025-04-29:
The `writeJointCommand` function is used for high-frequency control such as in [this example](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/examples/full_driver.cpp) or the [ROS driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver).

When using this interface the responsibility to send valid, reachable goals is at the caller's side.