UniversalRobots/Universal_Robots_Client_Library #322: Clarify PolyScope X support

urfeex, 2025-05-15:
Throughout the docs we have marked PolyScope X support preliminary, e.g. [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/polyscope_compatibility.html) or [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/setup/robot_setup.html).

From my understanding, the ExternalControl URCapX is now in a state where we could remove the beta statements, would you agree @urrsk @urmahp?

urrsk, 2025-05-20:
I will say we need to have the External control for Polyscope X to support multiple external control nodes in one program before we can remove the beta flag

urmahp, 2025-05-21:
I wouldn't consider the URCap as under development any longer, so I would either delete the last comment or just write, that you would need a different URCap for PolyScopeX.  

[image]

Do we consider to make any breaking changes to the URCap? Because if we are not I would consider removing the beta flag and then just list the features that isn't supported yet.