UniversalRobots/Universal_Robots_Client_Library #328: Failed to bind socket for port xxxx to address. Reason: Address already in useRetrying in 1 seconds

zkeyg, 2025-05-21:
### Affected Client Library version(s)

2.0.0

### What combination of platform is the Client Library running on.

Linux without realtime patch, Docker, Linux

### How did you install the Client Library

Build the library from source

### Which robot platform is the library connected to.

Real robot, UR E-series robot

### Robot SW / URSim version(s)

polyscope 5.21.x

### How are you using the Client Library

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
[UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds

## Issue details
Using UR-ROS2 driver connected to a UR20 robot using external control cap and UR client library. It appears that some times I'm encountering an issue where the TCP server fails to bind to port 50002 with the error Address already in use, even though no active process is using the port (confirmed via lsof -i :50002). I have a short script checking if port is being used or occupied before the driver is launched. Nothing is detected. My suspicion is that the port is likely held in TIME_WAIT state by the OS, causing the bind() call to fail.

## Steps to Reproduce
Start the TCP server on port 50002.
Stop the server.
Restart the server quickly (within a few seconds).
Failed to bind socket for port 50002 to address. Reason: Address already in use

## Expected Behavior

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
I am wondering if adding a flag in the TCP bind function will be useful. Ideally, the TCP server should set SO_REUSEADDR before calling bind() to avoid this issue in the future.https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/ccc2e5ab02588b2b495390d3a466bce602d184cc/src/comm/tcp_server.cpp#L132

```
int opt = 1;
setsockopt(listen_fd_, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));
```

### Relevant log output

```shell
948
2025-05-17T00:11:23.113Z
[ros2-9] [ur_ros2_control_node-1] [WARN] [1747440683.112865892] [UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds
949
2025-05-17T00:11:24.114Z
[ros2-9] [ur_ros2_control_node-1] [WARN] [1747440684.112995375] [UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds
950
2025-05-17T00:11:25.113Z
[ros2-9] [ur_ros2_control_node-1] [WARN] [1747440685.113131975] [UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds
951
2025-05-17T00:11:26.113Z
[ros2-9] [ur_ros2_control_node-1] [WARN] [1747440686.113244548] [UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds
952
2025-05-17T00:11:27.113Z
[ros2-9] [ur_ros2_control_node-1] [WARN] [1747440687.113353851] [UR_Client_Library:]: Failed to bind socket for port 50002 to address. Reason: Address already in useRetrying in 1 seconds
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-05-22:
I have experienced this problem in the past. On some systems I was able to reproduce this issue somewhat reliable, on others not at all, so I never followed it too much. Though, when writing this I remember, that we even have a sleep in our CI between running the examples because of exactly that reason...

Setting `SO_REUSEADDR` seems like a reasonable approach to this problem, thank you for pointing us into that direction!

zkeyg, 2025-05-22:
Appreciate it! I tested on my own UR20 but have not tested on other platforms.