UniversalRobots/Universal_Robots_Client_Library #352: [FEATURE NAME] Inquiry Regarding RTDE Control Interface API for Direct Torque Control

kwongeunwoo, 2025-07-05:
### Feature summary

Hello,

I am a student currently attempting direct torque control on a UR5e robot. I am working in the PolyScope 5.22.0 beta environment.
I am using the RTDE Control Interface API to send torque commands to the robot.

I have written a simple example code that sets all joint torques to zero using #include <ur_rtde/rtde_control_interface.h>.
However, when I run the code, the robot vibrates excessively even with the slightest touch.
I am reaching out to ask if there are any solutions or advice regarding this issue.

Below is the code I have written: 

Thank you for your attention.
==================================================================================
#include <ur_rtde/rtde_control_interface.h>
#include <ur_rtde/rtde_receive_interface.h>
#include <vector>
#include <thread>
#include <atomic>
#include <csignal>
#include <chrono>
#include <iostream>

using namespace ur_rtde;

std::atomic<bool> running(true);

void signal_handler(int signal) {
    if (signal == SIGINT) {
        std::cout << "\nCtrl+C pressed, exiting..." << std::endl;
        running = false;
    }
}

int main(int argc, char* argv[]) {
    std::cout << "Starting Torque Zero Command (C++)" << std::endl;

    RTDEControlInterface rtde_control("192.168.235.5", 500.0, RTDEControlInterface::FLAG_USE_EXT_UR_CAP, 50002);
    RTDEReceiveInterface rtde_receive("192.168.235.5", 500.0);

    if (!rtde_receive.isConnected()) {
        std::cerr << "[ERROR] RTDEReceiveInterface not connected!" << std::endl;
        return 1;
    }
    if (!rtde_control.isConnected()) {
        std::cerr << "[ERROR] RTDEControlInterface not connected!" << std::endl;
        return 1;
    }
    std::cout << "Successfully connected to robot!\n";

    std::signal(SIGINT, signal_handler);

    const double dt = 1.0 / 500.0; 

    std::cout << "Start torque zero command loop (Ctrl+C to exit)\n";
    while (running) {
        auto t_start = rtde_control.initPeriod();

        std::vector<double> tau_zero(6, 0.0); 
        rtde_control.torqueCommand(tau_zero, true);

        rtde_control.waitPeriod(t_start); 
    }

    rtde_control.stopScript();
    std::cout << "Torque zero finished.\n";
    return 0;
}
==================================================================================

kwongeunwoo, 2025-07-07:
Thank you very much for your response.

In that case, could you kindly let me know where I can ask questions related to ur_rtde? I appreciate your help!

urfeex, 2025-07-07:
Hm, I was not aware that the beta has a 5.22.0 version number.  In that case we would need to revert #348 temporarily.

For the linked PR you can simply remove the check in https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/353/files#diff-ce13397693db9c9b0924b920a7aea8f76da3b768945c8e31d3c19319c391cca4R66-R77