UniversalRobots/Universal_Robots_Client_Library #360: behavior of commandIsInRemoteControl() while using simulation

JohnN193, 2025-07-10:
### Affected Client Library version(s)

2.1.0

### What combination of platform is the Client Library running on.

Linux

### How did you install the Client Library

Build the library from source

### Which robot platform is the library connected to.

URSim in linux

### Robot SW / URSim version(s)

5.21.3

### How are you using the Client Library

Headless without using the teach pendant

### Issue details

# Summary
When using the DashboardClient with ur_sim `commandIsInRemoteControl`'s behavior is unclear
## Issue details
When using the DashboardClient with ur_sim `commandIsInRemoteControl` appears to always return `false`, or in manual mode. We were wondering if this was expected behavior that ur_sim will always return this, and what the reasoning for this is. 

I think the reason for this is because ur_sim already behaves like its in local mode, where you can move the arm from the ui, and that ur_sim skips the step where the arm will block connections while configured for manual/local mode

# Workaround Suggestion
It seems like its safe to assume that sim will always return commandIsInRemoteControl=false

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-07-11:
I cannot reproduce this. I've stripped down the dashboard example to only make that call and had it running against a URSim 5.21.3. See the video below.

https://github.com/user-attachments/assets/73f1ae5f-a3a8-4d27-9cba-4001c6f6ef4a

However, remote control mode in URSim is indeed a bit special. There is (to my knowledge) no real difference between being in remote control mode or not, as you have already mentioned.

JohnN193, 2025-07-24:
Sorry forgot to respond!

at the time, I was testing with the ur5e sim, and the local/remote toggle wasn't appearing on URSim. using the ur20 sim I do see the local/remote toggle.

[image]

I was looking for a way to determine if we were using a URSim arm or not, and while being able to control the arm during local mode will give me the answer.

Are there any other apis on the DashboardClient or any commands we can send to determine this?

urfeex, 2025-07-25:
I don't know of a direct, reliable way to tell that. A workaround might be to leverage the `get serial number` dashboard call. To my knowledge, all URSim robots have serial numbers ending in ``99999`. But that's what I meant initially with "reliable": There's no guarantee that this will change at some point.

I'll mark this issue as closed, as the initial question was answered I think,

closed 2025-07-25 by urfeex