UniversalRobots/Universal_Robots_Client_Library #367: Issue name

becketps, 2025-08-02:
### Affected Client Library version(s)

All ARM based Systems

### What combination of platform is the Client Library running on.

Linux

### How did you install the Client Library

Build the library from source

### Which robot platform is the library connected to.

URSim in docker

### Robot SW / URSim version(s)

10.9.0

### How are you using the Client Library

Others

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*

I wanted to start the polyscope X docker container with this script 

https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/scripts/start_ursim.sh

I ran ./start_ursim.sh -v 10.9.0 -m ur15 to install and running the docker container on an Jetson AGX Orin PC. 

It pulled the docker image but it did not pull everything and  the service does not start. Can you please help me finding a solution for an ARM System like the AI Accelerator? 

I also installed Docker engine which works fine for other docker container. 

### Relevant log output

```shell
./start_ursim.sh -v 10.9.0 -m ur15
ROBOT_MODEL: ur15
ROBOT_SERIES: polyscopex
URSIM_VERSION: 10.9.0
Unable to find image 'universalrobots/ursim_polyscopex:0.14.47' locally
0.14.47: Pulling from universalrobots/ursim_polyscopex
6e771e15690e: Pulling fs layer 
e908642e05b5: Pulling fs layer 
4f4fb700ef54: Pulling fs layer 
921ebed4db9f: Waiting 
17e39b077a4c: Waiting 
3a55093f8762: Waiting 
4d4a4a535b8c: Pull complete 
8aa4121a3e10: Pull complete 
9d159dcc3a2a: Pull complete 
57c2b2213d30: Pull complete 
988965604792: Pull complete 
b76eefe96f3a: Pull complete 
78bf806e19af: Pull complete 
5378ef726913: Pull complete 
8f09d35ff668: Pull complete 
d1ad1a9ed82e: Pull complete 
571a4501afd3: Pull complete 
b606ea87b5dc: Pull complete 
fa205a4f970a: Pull complete 
712f76785bd6: Pull complete 
e500ebfb3730: Pull complete 
58dde21864c5: Pull complete 
dd49c5a84df6: Pull complete 
9cdbed2e3b26: Pull complete 
e869071b600d: Pull complete 
c74bb1206395: Pull complete 
6e8170feb756: Pull complete 
443cbf78b27d: Pull complete 
e4b6cdb139ef: Pull complete 
Digest: sha256:9904a17cbaf09c4174127afc8fd00c9123bd3391765ef98c47e61212fa63d400
Status: Downloaded newer image for universalrobots/ursim_polyscopex:0.14.47
08b61f8ffbad645830e3df95fdf123fc16202378d89e96dd7e5b7bc82c45ed93
Downloading External Control URCapX version 1.0.0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 21.1M  100 21.1M    0     0  8612k      0  0:00:02  0:00:02 --:--:-- 12.1M
Starting URSim. Waiting for UrService to be up.............................................................................................................................................................................................................................
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-08-04:
Thank you for bringing that up. The [image itself](https://hub.docker.com/r/universalrobots/ursim_polyscopex) expects an env variable specifying the host architecture if it's not AMD64. We're not supporting that in our start script, yet.

A first hacky workaround would be to modify the script for you locally adding `-e HOST_ARCH=arm64` [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/79672841a934e16a2ea84bc3eba6ca1720fd6cc5/scripts/start_ursim.sh#L456).

Could you please check whether that works for you? Then we could add a flag for this. We could also setup automatic parsing of the output of `arch` with falling back to user input if a) the command isn't available or b) the output is an unknown string (as there are more arch strings out there than x86_64 and aarch64)

becketps, 2025-08-04:
unfortunately both methods does not work.

On the starting skript I added the -e HOST_ARCH=arm64 and executed it. But it did not pull everything and does not continue. 

The second method which failed was

docker run --rm -it \
  --platform linux/amd64 \
  --add-host "host.docker.internal:host-gateway" \
  --env HOST_ARCH=arm64 \
  --network bridge \
  --privileged \
  -p 80:80 \
  universalrobots/ursim_polyscopex:0.15.78
exec /usr/local/bin/run-docker.sh: exec format error

In general no ip was shown where the ursim runs on.

urfeex, 2025-08-05:
The PolyScope simulator apparently only works on AMD64 and **MAC** ARM64.

becketps, 2025-09-10:
Thanks for the hint but I connected the Jetson via Ethernetcable to the PC. On the PC I did the process you mention above. Then I tried to reach the ur_sim from the jetson but did not succeed yet. 

The building process of the ros2 driver on the jetson also has some bugs.